makersmakingchange/LipSync #4: Berzerk Mode

jakemcivor, 2024-03-06:


jakemcivor, 2024-04-05:
This issue seems to be related or made worse by having a shortened mouthpiece. We are investigating the possibility of a saturated filter contributing to the issue as well.

jakemcivor, 2024-08-25:
Got a report that this is still occurring occasionally. The user rarely uses the puff function, but this behaviour still occurs. Changing the filter or modifying the mouthpiece length doesn't seem to resolve the issue.

jversloot, 2024-10-09:
Got a report of a problem that I think may be related, this is the message:

"I’m having some issues with my new Lipsync. It freezes up randomly, sometimes every few days sometimes it can freeze up a couple times in a day.
Normally when I puff the first feedback light comes on. When it freezes up the  3rd feedback lights comes on. The only way to fix the problem is to restart my Ipad.
I tried to do a calibration and a microcontroller reset, still the same issue."

jakemcivor, 2025-01-30:
We've tracked down the source of this error to a section of the code that reset the sip and puff baseline when a large change in ambient pressure occurred. We were able to reproduce 'berzerk' mode by applying a sip to the mouthpiece, holding it closed (with a negative pressure still on the tube pressure sensor) and blowing on the ambient pressure sensor.

This caused the sip and puff baseline to be reset, which resulted in a constant sip being registered, that wouldn't be reset unless the power was cycled.

closed 2025-04-01 by jversloot