makersmakingchange/LipSync #31: LipSync doesn't travel well

bradleywell, 2024-11-25:
Some clinicians say that the LipSync does not travel well and need recalibration and re-setup at every location. They have asked for some sort of travel cover that covers the gimbal and prevents it from being jostled or bumped in transit. Following up with them for more detail.

bradleywell, 2025-02-10:
Following up with the clinician and got the following feedback. Sounds like an underlying bug and not an issue with travel itself.

"What I had found with the LipSync is that something about transporting it is causing errors, and a full factory reset seems to be required to get it functioning again. This happened when transporting it for a demo at a university, and also happened to a client who moved house recently and found that it would be detected by the computer after the move but would not display a cursor or seemed to have any control over mouse movement. This client had customized the sip/puff strength and duration to match his wheelchair, so was frustrated at having to do the factory reset to get it working again. He wished there was an easy readout screen showing any changes made from default settings that he could have taken a picture of for reference before resetting."

jakemcivor, 2025-02-10:
I suspect the root cause of this is probably automatically switching to Bluetooth (i.e., https://github.com/makersmakingchange/LipSync/issues/23). The 4.1 firmware update will resolve that. In the meantime, changing the operating mode back to USB Mouse instead of doing a Factory Reset will enable the user to resolve the problem without losing their settings. 

Will add a corresponding feature request for the ability to more easily restore user settings, although that would be challenging with current hardware.

bradleywell, 2025-02-10:
I've passed this along to the clinician and asked her if she can test this and try changing it to USB mouse and see if that fixes it.