makersmakingchange/LipSync #37: TLV493D-A1B6 Library is deprecated

jakemcivor, 2025-01-09:
The library for the magnetic sensor is deprecated.

TLV493D-A1B6 
This library is deprecated and no longer maintained. Please use our [XENSIV™ 3D Magnetic Sensor TLx493D Arduino Library](https://github.com/Infineon/arduino-xensiv-3d-magnetic-sensor-tlx493d) instead.

jversloot, 2025-07-24:
Library seems to be removed from Arduino Library manager now, so new makers will not be able to add it. Will need to investigate a solution for this.

jakemcivor, 2025-11-13:
The Maker Guide was updated to include instructions to manually download and add the deprecated library using the zip file (https://github.com/Infineon/TLV493D-A1B6-3DMagnetic-Sensor/archive/refs/heads/master.zip).

The longer term solution is to update the code to utilize the new library (https://github.com/Infineon/arduino-xensiv-3d-magnetic-sensor-tlx493d)